STMicroelectronics-CentralLabs/ST_Drone_FCU_F401 #1: SWD not working

KiranKanchi, 2017-12-11:
The device is not working with ST LINK V2 provided in the STM32F4 Discovery.
does the device support DFU ?
Is the USB  micro port configured as USB device on factory firmware ?

KiranKanchi, 2017-12-13:
never mind , i got it.

gmessina72, 2017-12-13:
Hi Kiran,
could you explain what was the problem?

Thanks

KiranKanchi, 2017-12-13:
The VDD at SWD pin was only 1.8V on STM32F4, later used 3v pin . also olimex connector needs a solder bridge for SCK.
i wish there were more instructions on user manual.

KiranKanchi, 2017-12-13:
[image]
[image]

the project worked on ac6 but not keil. there seem to be some missing definition. when is the next estimated firmware update ?

teomaras76, 2017-12-14:
Hi Kiran,
Did you plug in the battery before accessing with SWD?
While debugging you need to have the battery connected otherwise 3.3V drops, and better to have also micro USB so your battery will not be discharged. It's written in the documentation but perhaps it was not enough highlighted. Later it will be published also a Getting started document.

Matteo

KiranKanchi, 2017-12-14:
Yeah, i got SWD access . Olimex connector needs solder bridge, i wish there
was a schematic for SWD.
The bluecon firmware has 73 errors and 9 linking errors on keil. Although
no errors on AC6.
I prefer keil, a document on using an external ESC would be great too.

gmessina72, 2017-12-14:
Hi Kiran,
the first version of documentation will be available very soon (maybe today) I'll keep you informed.

teomaras76, 2017-12-14:
For Olimex connector sounds strange to me, let me check. My Olimex connector worked well without any modification.

For Keil I will check, which version are you using?

For the external ESC it's already working, even if not widely tested like the DC Motor. You can find a define in motor.h to change motor type. Once you have changed to external ESC configuration you may need to fine tune the PWM ton pulse duration for arming the ESC, since every ESC has a little different value (usually it's 1msec to arm and 1.1msec to start the motor). Also this point will be in the getting started documentation.

In next days SW User Manual will be published, you may find in it explanation also for the ESC.

KiranKanchi, 2017-12-14:
That would be incredible !
Thank you :-)
Hey would you recommend any textbook to understand the flight controller
logic?
I see a lot of 3D mathematics and feedback control systems. Any hint would
be awesome

gmessina72, 2017-12-14:
In term of textbook I can give you some link to publications, already present into the documentation.

Quaternion-Based Kalman Filter for AHRS Using an Adaptive-Step Gradient Descent Algorithm: http://journals.sagepub.com/doi/full/10.5772/61313 
Filtro Complementare – Metodo di Mahony: http://ieeexplore.ieee.org/document/4608934/
IMU Data Fusing: Complementary, Kalman, and Mahony Filter:http://www.olliw.eu/2013/imu-data-fusing/ 
Quadcopter Pid Explained: https://oscarliang.com/quadcopter-pid-explained-tuning/ 
PID Tuning and Quadcopter Flight Behaviour: https://oscarliang.com/understanding-pid-for-quadcopter-rc-flight/

KiranKanchi, 2017-12-14:
MDK5. For Bluetooth control, no change in sw necessary?
Just compile and program?

teomaras76, 2017-12-14:
Yes only SWD. For Bluetootooth Low Energy control no need to change FW, just compile and program, HOWEVER please notice that it's still a beta version and to be considered work in progress. We are going to update both app and FW for STM32 in next weeks. At this moment you can achieve the best drone performance and control with the external remocon, in case you may have one.

teomaras76, 2017-12-14:
Also please notice that if you are going to use it with external ESC for outdoor use, the maximum distance for BLE connection is about 10-15 meters.

KiranKanchi, 2017-12-14:
I will start with brushed motors first, i nearly ripped my arms with a
2200kv motor once with naze32.
Hey what about calibration?
Can i use stock calibration for brushed motors ? And i guess calibration is
needed for external esc and large motors.
I am also going to use a commercial chassis with 8520 motors for now

teomaras76, 2017-12-14:
There are 2 kinds of calibration:
- sensor calibration: in actual FW you need to perform every time you switch on the FCU. You need to put the drone on a flat surface and the sticks in a certain position (please refer to user manual for it) to perform it
- ESC calibration: you need to insert the values for calibration manually in the FW (motor.h), also this point is explained in the SW manual.

SW User Manual link:
http://www.st.com/content/ccc/resource/technical/document/user_manual/group0/b5/30/a9/84/6f/62/42/a3/DM00456644/files/DM00456644.pdf/jcr:content/translations/en.DM00456644.pdf

supermosfet, 2017-12-18:
Kiran,
I was not able to flash firmware on the FCU001. I used an STLINK from my Nucleo board using Systems Workbench nor Atollic. My computers could not process STLink's response.
Can you please tell me more about the two issues you discovered.

1. The VDD at SWD pin was only 1.8V on STM32F4, later used 3v pin. What  exactly did you do to make it 3v?
2. also olimex connector needs a solder bridge for SCK. What pins needs to be soldered? 
Here is the schematics I used [https://www.olimex.com/Products/ARM/JTAG/ARM-JTAG-20-10/resources/ARM-JTAG-20-10-schematic.pdf](url)

supermosfet, 2017-12-18:
What should be the behavior of the board to indicate it has been flashed successfully?

KiranKanchi, 2017-12-18:
Hi, 
Try using supply for SWD from VDD of nucleo MCU
just use one gnd pin in the olimex anyway 2 to 20 are shorted  and one VDD
SCK,SWD is sufficient.
Use a multimeter to make continuity test, that worked for me. And yes when you search for sck on olimex through continuity test you may not get a hit.

KiranKanchi, 2017-12-18:
I needed a solder bridge for sck, my olimex could have been defective .
The fcu does not have any lights to indicate flash however my stm32f4 has a nice led that toggles between red and green on successful debug link . All the lights on fcu will turn off during flash

KiranKanchi, 2017-12-18:
Try ST link utility first , try to read flash data so it's irrespective of IDE .

KiranKanchi, 2017-12-18:
LD1 is continuously lit and LD2 blinks on successful firmware flash

KiranKanchi, 2017-12-18:
Short R3 pads on olimex if you dont get a hit for sck on continuity test

teomaras76, 2017-12-18:
Please download newest version of FW project for BLE. New PID parameters solve an issue of drift at lift off.

supermosfet, 2017-12-18:
Thank you, guys. 
I was able to solve my issue with being unable to flash the FCU by connecting a 3.7V battery. Kiran, the  LEDs are flashing as you indicated.

KiranKanchi, 2017-12-18:
I'm glad to hear that :-)
So what do you plan with the FCU?
DC motors or BLDC with ESC ?
I am confused as of now, looking for 8520 motors at aliexpress.

supermosfet, 2017-12-18:
Kiran,
I am repurposing my old toy drone with brushed motors. I also bought a Spectrum 4 Channel radio. The next step would be upgrading the drone to use both Bluetooth and radio.

gmessina72, 2017-12-19:
Updated FW, now KEIL project compile. Must be reviewed to remove warnings.

closed 2017-12-19 by gmessina72