STMicroelectronics-CentralLabs/ST_Drone_FCU_F401 #2: App - Motors not armed

kylix34, 2017-12-15:
even if you arm the motors, in the app GUI you always see the "Not armed" red button.
It will be also useful to have an indicator of the battery voltage level.

teomaras76, 2017-12-16:
Yes you are right, thanks a lot for your comment. BLE app is still in beta version and now we are focusing on fine tuning the Fw ( both on FCU and App side) to can fly the drone with BLE app with performance as much as possible to external mechanical remocon (that as now is the best for agility and manouvrability performance. We will release next week a new version with better performance and better stability for BLE connection.

We have already in plan in following weeks (or beg of next year) to enable in the FCU FW the feature to send several additional information:
- sensors value
- battery value
- arming status
-....

teomaras76, 2017-12-18:
Please download newest version of FW project for BLE. New PID parameters solve an issue of drift at lift off.

KiranKanchi, 2017-12-18:
source code for app will be published ?

teomaras76, 2017-12-18:
Yes it will be published as soon as the revision will be a stable one, now it'a still in beta and under debugging so some parts may change.

KiranKanchi, 2017-12-18:
Awesome :-) can we expect a few more video tutorials on YouTube?

teomaras76, 2017-12-18:
Yes next video will be more focused on the FW project and use of the App

kylix34, 2017-12-18:
Firmware updated but it still has a slight drift at lift off. I might need to better position the battery.

KiranKanchi, 2017-12-18:
Kylix, are you using BLDC or DC motor setup ?

kylix34, 2017-12-18:
DC motors (the ones from the kit). It would be nice to have a forum or something like this to talk, instead of Github.

KiranKanchi, 2017-12-18:
True that, 8520 motors right ? Propeller 65mm ?
Could post a pic of your build please ?
Using Bluetooth app to control?

kylix34, 2017-12-18:
Yes, 8520 motors and 65 mm propellers. I'm using my Android phone to control it but it's not easy :)
[image]
[image]
[image]
[image]

KiranKanchi, 2017-12-18:
Wow, sweet build <3 did you print props too ?

kylix34, 2017-12-18:
Nope, the propellers are standard ...

teomaras76, 2017-12-18:
Kylix, nice and ver clean built!
Yes we are going to have also a dedicated forum for it beginning of next year, now we are preparing the page.
A small drift at the lift off it's normal (I think you already did the sensor calibration procedure) and is related to the so-called "ground effect": when the drone is too close to ground, altitude less than 2-3 time size of diameter of drone, the air moved from propeller is not circulating smoothly and makes the drone control not easy.

teomaras76, 2017-12-18:
Do you notice some drift also when you take the drone at more than 1m altitude?

kylix34, 2017-12-19:
Hello Teo!
You are right, I'm noticing the drift only when the drone is lower than 20-30 cm to the ground.

teomaras76, 2017-12-19:
Ok this is normal.  There is a ground effect, against with the drone is struggling to keep stable asset, and since with propeller guard is a little heavy it struggles for stability and there is higher vibration. After that altitude it becomes easier for the drone to have stable hovering.

teomaras76, 2017-12-19:
At lift off you need to manually compensate a little, with mechanical remocon is easier and you can have a smooth lift off.

teomaras76, 2017-12-19:
With BLE app actually you are not so fast to give commands to compensate so it's more difficult to lift off. When you can pass about 30cm threshold everything becomes easier.

kylix34, 2017-12-19:
Thanks! Could you advice on a remocon/receiver model?

teomaras76, 2017-12-19:
Sorry officially I cannot recommend a specific one... What I can say that Walkera Devo 7E with Deviation FW is cheap and I am using it for testing the minidrone successfully. Also small size and easy to carry on.

closed 2017-12-19 by kylix34

teomaras76, 2018-04-11:
Hi !
Last week we opened a new community specific for Drone on ST website, please kindly have a look here:
https://community.st.com/community/drone-zone
It would be used also as a generic forum for Q&A both on ST FCU or ST ESC demo boards.
Also it would be very appreciated if you may share with other community users any photo or video of the minidrone you built with the ST FCU.
 
Matteo