STMicroelectronics-CentralLabs/ST_Drone_FCU_F401 #4: HW Component failure

KiranKanchi, 2017-12-26:
Hello, my magnetometer does not initialize .
It was working fine, now it stopped working spontaneously.
Please direct me to hardware support.

teomaras76, 2017-12-28:
Hi, it sounds a little strange that it does stop to work, may you give more details about it?
Is it failing during the initialization process? Can you read RAW data from it?

KiranKanchi, 2017-12-28:
I soldered headers pins to it and flashed with latest version .the two leds
stay lit. Because when sensor init routine comes up the control gets stuck
in a while 1 loop.
I had to comment of while(1) for each sensor one of after the other to
identify the issue .
Finally when i disabled magnetometer led started blinking and found the
Bluetooth device.

teomaras76, 2018-01-02:
Had you the same issue with other versions of the FW project? If you can try the first official version (the one without BLE) may help to discriminate if real HW issue or some FW bug of latest version.
In the main you can find the initialization of all the sensors (please double check an eventually uncomment the part initializing the magnetometer) and you can also PRINTF magnetometer raw data on UART for check (you can find some commented parts with PRINTF left for debug).
The boards are tested in production, also the correct initialization of magnetometer is tested, this is why better to make sure again it's a HW issue.

teomaras76, 2018-01-02:
I read you had same issue with also previous version, please kindly confirm the following:
- test at first with first version of FW without BLE
- please confirm if the issue appeared from very first beginning when you used the board for the first time, or if the problem appeared later, eventually after some drone crash

KiranKanchi, 2018-01-08:
hello , i tried with the very original firmware too.
if (BSP_MAGNETO_Init( LIS2MDL_M_0, &LIS2MDL_M_0_handle ) != COMPONENT_OK)
while(1); 
the control does not flow out of while(1); loop. i have checked many times.
i still didnt fly with fcu to crash too, i dont understand what is wrong with sensor.

closed 2018-01-12 by KiranKanchi