STMicroelectronics-CentralLabs/ST_Drone_FCU_F401 #7: BLE Remocon Firmware + app v11

ziogianni, 2018-01-18:
Hello I have an issue with the android app, as soon as I arm the drone using the proper command sequence, the DC motors start to spin but the application loses the connection with the STEVAL board.

Then I'm forced to reset the board and start it over.

Any help?

teomaras76, 2018-01-19:
Hi, can you make again the same test with debugger connected  or check LED status and verify if a MCU reset is occurring?
It may be that when spinning the motors a sudden drop of voltage on Vbatt bus may reset the MCU.

teomaras76, 2018-01-20:
If it's an issue of voltage drop it should be related to battery used. A different battery with higher discharge rate should solve situation, or an additional capacitor on the Vbatt

ziogianni, 2018-01-20:
Hello Matteo, actually I have no battery connected yet, I was using the
power supplied by the ST Link just for testing purposes. The LED placed in
the middle of the board which turns green when the Bluetooth is connected,
after the arming procedute turns amber. Do you think could be tied to the
low current provided with such USB interface?

teomaras76, 2018-01-21:
Hi Gianni,
The power given by the USB is not enough and there will be a drop of 3.3V voltage given to STM32. If you look at schematics the USB will give power to the battery charger that will charge battery but with no battery inserted it will not give enough current. Also during debugging phase, with or without motor spinning, battery must always be connected.

ziogianni, 2018-01-21:
Ok Matteo, thanks for the heads-up.

As soon as my battery arrives I'll update you on this issue. But I'm pretty
confident, it's due to the low current drawn from the USB interface. That's
what I guessed and now I got a corfimation

inishchith, 2018-01-27:
@teomaras76 I'm from AboutCode org , can i know the category of the license (BSD or proprietary).
couldn't find your email id .

teomaras76, 2018-02-01:
@inishchith you may find the description of license in the file LICENSE at top page, is the information enough for you?

teomaras76, 2018-02-01:
@ziogianni please may you kindly comment if you have solved your issue so I may close this issue post?

inishchith, 2018-02-01:
@teomaras76 Yes , thanks !

ziogianni, 2018-02-01:
Matteo actually I haven't had any chance to try yet. I'll let you know

closed 2018-02-05 by teomaras76