STMicroelectronics-CentralLabs/ST_Drone_FCU_F401 #9: Turning off the drone

ziogianni, 2018-02-28:
Hello how can I turn off the drone without disconnecting the battery every time? Have I overlooked any option in the appDrone?

teomaras76, 2018-03-02:
Hi, sorry what do you mean by "turning off" the board? Please kindly explain more in details. The battery is connected to the Vbus on the board by HW, and also the STM32 is always connected, there is no HW switch option on the board.

You may turn off the LED by FW, but be careful that STM32 will continue to drain current from battery and you may risk to discharge it completely.
Disconnecting the battery is the safest operation in most of drones in the market as well, whenever there is not an HW switch.

ziogianni, 2018-03-02:
Hi Matteo, yes I meant to say to turn off the Steval board, but you've
already answered. Thanks

teomaras76, 2018-04-11:
Hi !
Last week we opened a new community specific for Drone on ST website, please kindly have a look here:
https://community.st.com/community/drone-zone
It would be used also as a generic forum for Q&A both on ST FCU or ST ESC demo boards.
Also it would be very appreciated if you may share with other community users any photo or video of the minidrone you built with the ST FCU.
 
Matteo