STMicroelectronics-CentralLabs/ST_Drone_FCU_F401 #10: I miss official release binaries

nelup305, 2018-04-04:
Hello. I bought STEVAL-FCU001V1 kit. When I connected the USB power, it looked that it was working, but I didn't succeeded with Bluetooth connection through Android application. I compile the project using SW4STM32, load the firmware (SWD), but it doesn't work right (both red LEDs blinks quickly - I lose debugging always around sensors init).

I want to load origin FW to check that board is still Ok. But I didn't find any binary release available. Is it possible to download origin binary, which is loaded during production?

Thanks.
Nelup

nelup305, 2018-04-04:
I found the problem. The board doesn't work right from USB. The battery or external supply is needed.

Nelup

closed 2018-04-04 by nelup305

teomaras76, 2018-04-05:
Yes correct the board is designed to work always with the battery connected, and it's suggested to connect also USB cable during debug to make sure the battery doesn't discharge.

teomaras76, 2018-04-11:
Hi !
Last week we opened a new community specific for Drone on ST website, please kindly have a look here:
https://community.st.com/community/drone-zone
It would be used also as a generic forum for Q&A both on ST FCU or ST ESC demo boards.
Also it would be very appreciated if you may share with other community users any photo or video of the minidrone you built with the ST FCU.
 
Matteo