STMicroelectronics-CentralLabs/ST_Drone_FCU_F401 #11: Can't find Drone via BLE

ungtu3012, 2018-04-15:
I uploaded ***Official release with BLE Remocon - 170318*** to board via STLink with TrueSTUDIO. Then LED 1 and LED 2 off. (I toggled RST button). But can't find Drone in Android app.

[image]

dinhvankiet124, 2018-04-17:
Thank you! This work for me :3

closed 2018-04-18 by ungtu3012

dinhvankiet124, 2018-04-18:
Hi, my friend ungtu3012 confirm that worked. But when I chose Target > Program & Start, two LED at the top were blinking once and stop immediately. Then the android app can't find them.

Android app v29

teomaras76, 2018-04-23:
Hi,
it's a very strange beahviour, if you download exactly same binary file ungtu3012 used (that I uploaded last week) it should work. Are you sure that battery is fully charged to 4.2V?
Even if the app is not yet connected to the FCU one LED should be still blinking.
I made simple video (not yet official one) at this link, please refer to it:
https://www.youtube.com/watch?v=RPNLA2R3kRo

have you downloaded the binary file using STLink Utility?
http://www.st.com/content/st_com/en/products/development-tools/software-development-tools/stm32-software-development-tools/stm32-programmers/stsw-link007.html