STMicroelectronics-CentralLabs/ST_Drone_FCU_F401 #12: Android app take off button missing?

dinhvankiet124, 2018-04-19:
I try to take off but not find take off button on Android app v29

dinhvankiet124, 2018-04-19:
[image]

messinajo, 2018-04-19:
Something seems to be wrong in your interface. 
This should be the correct one:

[image]

messinajo, 2018-04-19:
Did you use the apk or you use the source code?

dinhvankiet124, 2018-04-19:
I use apk! ì you click on the bar after a joint stick, you will get my images

messinajo, 2018-04-20:
If you have armed the motor you only need to move the left slider (joystick) up. (living the interface as I show you)

ungtu3012, 2018-04-20:
I already done it while "armed" button become green but nothing happen.
when i unplugged St-link V2 from board and even plugged battery into board, i couldn't connect BLE again.I and dingvankiet124 work together :")

Thanks for you support!!!

messinajo, 2018-04-20:
Can your refer if you do exactly the following steps:
1) Without anything connected to the board, plug the battery connector (battery fully charged)
2) The two leds swith on, one with fixed light and one blinking.
3) from the app, connect through bluetooth to the board
4)  calibrate sensors (first button from left, in the last raw)
5)  arm the motors
6) push the left joystick cursor up.

That's all.

Let me know.

dinhvankiet124, 2018-04-20:
1-2. There are 3 leds, 2 leds that you talk about not light. Only one led that far away on (with battery)

When disconnect STLINK, android app can't find board

closed 2018-04-22 by dinhvankiet124

teomaras76, 2018-04-23:
Hi, I made simple video (not yet ST official one but temporary), please refer to it for the moment for BLE operation:
https://www.youtube.com/watch?v=RPNLA2R3kRo