STMicroelectronics-CentralLabs/ST_Drone_FCU_F401 #14: Reading sensor values in AccGyroMagCharHandle using Bluez

eyewy, 2018-10-04:
I am trying to read the ACC+GYRO+MAG data stream using bluez. So I connect to the FCU:
pi@raspberrypi:~ $ sudo gatttool -b EF:2F:D0:24:0D:B2 -t random -I
[EF:2F:D0:24:0D:B2][LE]> connect
Attempting to connect to EF:2F:D0:24:0D:B2
Connection successful

I have put a printf in sensor_service.c to read the AccGyroMagCharHandle:
  ret = ACI_GATT_UPDATE_CHAR_VALUE(HWServW2STHandle, AccGyroMagCharHandle, 0, 2+3*3*2, buff);
  PRINTF("%u\n", AccGyroMagCharHandle);

From my PC serial terminal I can see successful connection, and that the handle value is 27 decimal (0x1b).
However, when I try to read this handle, I always get 1 line of constant data:
[EF:2F:D0:24:0D:B2][LE]> char-read-hnd 1b
Characteristic value/descriptor: 10 1c 00 1b c5 d5 a5 02 00 36 ac e1 11 01 00 00 00 e0 00 

I was hoping to get a stream of data, similar to sensortile as explained here:
http://iot.ee.ucla.edu/UCLA-STMicro/index.php/Home

Can you please hep me with the correct way of reading this data?
Thanks

teomaras76, 2018-10-24:
Hi, sorry I just noticed your msg now. Let me try and check since I never used Raspberry PI to check BLE data stream, I thinkl is a nice idea!

teomaras76, 2018-10-24:
Please can you let me know if you are using the following SDK on the Raspberry Pi?
https://github.com/STMicroelectronics-CentralLabs/BlueSTSDK_Python

eyewy, 2018-10-26:
Hi @teomaras76 ,

On Raspberry pi, I am not using any specific SDK, I am just using "gatttool"  command from BlueZ. Very similar to the official ST tutorial mentioned here: 
http://iot.ee.ucla.edu/UCLA-STMicro/index.php/Home

teomaras76, 2018-11-20:
Hi,
I'm trying to replicate your test with two different sets in order to find where is the issue:
SET 1) Raspberry PI 3B and Bluecoin evaluation board with FP-SNS-ALLMEMS1 FW package
SET 2) Raspberry PI 3B and FCU001 evaluation board with official Github latest FW release
I can connect successfull as you indicated in your description:
pi@raspberrypi:~ $ sudo gatttool -b **Board_MAC_ADDR** -t random -I
[EF:2F:D0:24:0D:B2][LE]> connect
Attempting to connect to **Board_MAC_ADDR**
Connection successful

Regarding next step is not very clear for me how you are proceeding. 
- Please may you post the exact gatttool commands you type on the RaspPi terminal after the "connect"?
- Before trying with the FCU board, have you tried same kind of operation to read the data with a different board and FW package like the Sensortile or Bluecoin? If yes can you post which FW package you used and if you applied any modification to the FW source code? (you wrote "**printf in sensor_service.c to read the AccGyroMagCharHandle**" but not clear for me if you tried same thing before on a sensortile or not)

Thanks

eyewy, 2018-11-21:
Hi @teomaras76 ,
I don't have SensorTile, however, I assume reading sensor data using Gatttool should be mostly similar in SensorTile and FCU boards. So I am following the official ST tutorial for SensorTile:
https://drive.google.com/file/d/1JVyw8-XIxEEnwGrHeDo7190aznLs3eTf/view
Section 3: Request Motion data using Gatttool
Unfortunately, there is not such tutorial for FCU, and I hope this thread will help in that direction.
By putting the printf in FCU firmware, I have found the "AccGyroMagCharHandle" to be equal to "1b".
So in the next step, I want to use Gatttool to get information from this handle.
In SensorTile, all we need to do is to enable the respective handle "handle #12" by issuing this command on RaspberryPi: "char-write-req 0012 0100" and the SensorTile starts streaming the data to PI. The FCU board is actually very similar, because I can see the motion data being transferred in live fashion from FCU to my Android phone. So there should be a way of getting the live motion data on raspberry pi as well, without modifying FCU firmware. So now, the questions are:
1. Is 0x001b the correct handle to get motion data (ACC, GYRO, MAG) on FCU?
2. How do we enable streaming from this handle, i.e., what value we need to write to this handle to enable streaming.

Thanks

teomaras76, 2018-12-06:
Hi sorry for taking time.
Here is the solution to your issue: correct handle for the ACCGYRMAG data is 0x001c so you need to write char-write-req 001d 0100 to see the RAW data scrolling.
To check the UUID for the available characteristics you can check uuid_ble_service.h in the FW project of the FCU.
I checked and it works with my RasPi, please try on your side too and let me know.

Matteo

eyewy, 2018-12-10:
Hi, thanks! this works.
So I understand how based on uuids in uuid_ble_service.h you got to the handle 1c, However, how did you figure out handle 1d, and the fact that you have to write 100 to it?

teomaras76, 2018-12-11:
Hi,
I checked at first which UUID is associated with ACC/GYR/MAG sensor data:

#define COPY_ACC_GYRO_MAG_W2ST_CHAR_UUID(uuid_struct)  COPY_UUID_128(uuid_struct,0x00,0xE0,0x00,0x00,0x00,0x01,0x11,0xe1,0xac,0x36,0x00,0x02,0xa5,0xd5,0xc5,0x1b)

After I checked which handle # is associated to this UUID using the gatttool. 100 must be written in order to enable that characteristics, 000 to disable (as in the UCLA guide). You can enable/disable in similar way using also BLE Scanner app on Android or iOS phones/tablets.
I hope it's clear now.

eyewy, 2018-12-13:
Thanks for your help.

closed 2018-12-13 by eyewy